thieu-b55/Slimme-meter-ESP32-V2 #4: PWM output

brandklijnstra, 2024-12-11:
Misschien heb ik het mis, of levert het geen probleem op. Als de PWM een puls uit stuurt, dan zal die niet in fase zijn met met de netspanning. In andere systemen wordt dit meestal opgelost dmv een trafo en een diode die na een ingang gaat. De opgaande flank van die ingang is dan begin van de PWM puls.

thieu-b55, 2024-12-13:
Het PWM signaal gaat naar een Solid State relais, pagina 32 van "Slimme meter ESP32 V2.pdf", hierin zou dit aanwezig moeten zijn, Voor mij werkt het in elk geval prima. Je kan dit zelf ook eens proberen door het verwarmingselement te vervangen door een goede ouderwetse gloeilamp en deze aan te sluiten op een solid state relais. De lichtsterkte van de lamp reageert mooi met het verhogen of verlagen van de duty cycle van het PWM signaal.

vriendelijke groeten,
thiue-b55

closed 2024-12-13 by thieu-b55

brandklijnstra, 2024-12-17:
Thieu,
dank voor je antwoord. Begin volgend jaar ga ik het een en ander proberen.
Aan de programma's heb ik de .h files toegevoegd. Waren prima te vinden.
Compileren leverde geen problemen op. Was eerst van plan om een ESP8266 te
gebruiken uit eigen voorraad, maar die heeft toch wat beperkingen. Ben ook
benieuwd hoe het gaat met de PWM. Wat ik bedoelde met het aansnijden van de
sinus door PWM, dit staat ook in de link beschreven hieronder. Ik heb zelf
een opleiding in de energietechniek gehad en hier in de praktijk ook mee te
maken gehad. Dit is natuurlijk geen kritiek op het ontwerp. Groet Brand

Zonne-energierouter om overproductie te beheren (doe-het-zelf) - F1ATB
<https://f1atb.fr/diy-solar-energy-router-to-manage-overproduction/>