thieu-b55/Slimme-meter-ESP32-V2 #7: Esp32 wroom blijft zichzelf herstarten

Srallie, 2025-04-03:
De esp blijft zich na een succesvolle firmware update met arduino ide herstarten. Ik kan even de webpagina bezoeken maar de verbinding verbreekt steeds,wifi led gaat uit op de esp en herstart zich.

thieu-b55, 2025-04-04:
Hoi Srallie,

Heb je de condensators parallel over de 5V geplaatst?
Kort na elkaar de webpagina proberen te updaten kan ook een probleem zijn.
Het vermogen van de 5V van de digitale meter is niet genoeg om de ESP32 te voeden, dit wordt opgevangen door condensators.
Probeer eventueel eens een externe 5V voeding.
Als dit niet het probleem is, heb ik helaas geen ander idee.

vriendelijke groeten,
thieu

Srallie, 2025-04-04:
Hoi,

Ik heb hem alleen ingestoken in de laptop usb poort en kijk mee met de seriële monitor van arduino,hier zie ik dat hij steeds opnieuw opstart. Ik heb hem nog niet fysiek gekoppeld aan de p1 poort. Kan dat hiermee te maken hebben dat hij nog geen inkomens signaal krijgt?

Gr Lars

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

thieu-b55, 2025-04-04:
Zonder inkomend signaal krijg je geen resultaten op de webpagina of sd kaart.
Ik veronderstel dat je USB poort niet genoeg vermogen kan leveren.
Om zeker te zijn, gewoon inpluggen in de P1 poort.

vriendelijke groeten,
thieu

thieu-b55, 2025-04-05:
*

closed 2025-04-05 by thieu-b55